WuMRC/drive #20: Changing Voltage of Pulse (Range)

itkwon, 2015-02-10:
Dear @tunjid ,

I looked through your library, and I figured out that you added some function to change the voltage of the pulse for measuring the impedance. Did you ever tried to change the "Range?" (and measure?)

According to the datasheet, the error introduces generally by difference between the internal impedance of AD5933 (depends on the "Range") and the impedance to measure. 

I think that, based on current impedance range, you can try 2Vpp or 0.2Vpp. (In addition, 0.2Vpp shows 0.15mA. It means that it works with potentiometers!)

If anyone have something to comment regarding my question, feel free to post here.
Thanks.

tunjid, 2015-02-10:
Hi Il Taek,

I haven't changed the range because I haven't needed to yet. I added the function over winter break as an exercise for myself in bit masking. I currently leave it in Range 1 as that's the default and it has the lowest output impedance, but I don't know how much that changes things since we're using the AD8606 as a buffer.

With your suggestion to using 2Vpp or 0.2Vpp for the potentiometer, could you explain more? I'm not very familiar with digital potentiometers so I don't know the effect choosing a particular range for a certain vpp value will have.

Thanks.

itkwon, 2015-02-11:
Thanks. 
1. The function was good for practice.
2. Yes, the AD8606 OP-AMP circuit reduces the output impedance.
3. Generally, digital potentiometer endures the 0.5mA output current as a maximum. (You have to look up the accurate value, and it depends on each chip.) Our first design uses 2Vpp pulse with 0.58mA current, which exceeds the maximum current. If we use 0.2Vpp pulse with 0.15mA current, we can use AD5933 with the potentiometer.
4. Besides, we might need to test how the result changes if we change the "range."

tunjid, 2015-02-11:
Oh I get what you mean now, I'll test the range function when I get in tomorrow.

itkwon, 2015-02-11:
Based on my recent test (with the circuit with MUX,) It works pretty well except more accumulating error. In other words, the error increases faster when the resistance (impedance) is farther from the calibration resistor. 
Based on the calculation of AN-1252, the current 332 Ohm feedback resistor can accommodate the resistance from 378 Ohm. In other words, the minimum limit is 378 Ohm. 
((3.3/2-0.2)*378/(0.173+3.3/2-0.173))

tunjid, 2015-02-13:
I see, when calibrating i'll use the maximum current with the lowest output impedance I can get away with. Thanks for the heads up!

closed 2015-04-20 by itkwon