WuMRC/drive #21: Calibration Issue

tunjid, 2015-02-18:
While the calibration sketch works as it should, there's the issue of not knowing exactly what resistor combination is engaged from the digital rheostats r1 and r2 at any one time.

The ideal case with each wiper position increasing linearly does not correspond with measured values from a multimeter or the LCR. The LCR at a fixed frequency (20 Hz) was then used to create a lookup table for each step for both rheostats. The Lookup tables are in the "Text Files" folder under Analysis/MATLAB. The data generated with this lookup tables is in the same folder with the name "Test".

After a brief test run, The values computed using the lookup table in the ideal model does not correspond to measured data by the AD5933. This is in contrast to previous tests with discrete r1, r2 and capacitor values measured by a multimeter.

I currently have two reasons this may be:

1. The lookup table generated from the LCR at 20 Hz is bad, and does not match the actual values from the rheostats.

2. As Il Taek mentioned in a previous issue, the AD5933 has to be used in Range 4 so the current injected is below 5 mA. This is because the rheostats are rated to an absolute maximum of 5 mA flowing through it continuously. Range 4 has an output impedance of about 600 Ohms, which is 3 times higher than Range 1 (The range used prior) whose output impedance is 200 Ohms. This should lead to a higher rate of accumulated errors as Il Taek pointed out.

The case may even be a combination of both reasons. As of right now, it needs to be ascertained what exactly the issue is before calibration can continue. My current plan is to somehow interface the LCR with the rheostats the same way I currently have the AD5933 and measure the rheostat values both with the LCR and AD5933.

The biggest challenge here is the software that can control both the LCR through the GPIB interface while simultaneously controlling the rheostats via i2c. LabView and Visual Basic are the current choices I have been able to narrow down so far, with my preference leaning towards Visual Basic.

Suggestions are very much welcome.

itkwon, 2015-02-18:
It is obvious, but did you tested the precise LCR meter with generic resistors? I have figured out the measurement has some difference to conclude that it is out of order. It could be my error, so I suggest you to test it.

tunjid, 2015-02-19:
Hi Il Taek,

I have used the LCR with generic resistors and it at single frequencies and it works fine, the base line I'm comparing it to here is a standard digital multimeter. The issues I've had with the LCR occur at frequency sweeps, especially around 10 KHz that suggest it needs to be recalibrated.

The issue I have now is while the digipot works fine, I don't know the actual resistance for each discrete step so I can't actually compare values measured by the AD5933 to a theoretical model, talk less of the LCR. 

The issue basically boils down to this, how do we create a very accurate lookup table for all 64 steps in both digitpots used in the experiment quickly?

itkwon, 2015-02-20:
Yes, I tested the LCR meter, and it is fine now. I know that you have a issue with the digital pot (potentiometer.)

According to the datasheet of AD5258 (http://www.analog.com/media/en/technical-documentation/data-sheets/AD5258.pdf), it has several kinds of this. It depends on end resistance (1k, 10k, 50k, 100k.)
What do you have? Please let me know the detailed model number.

[image]
Above is the equation of the resistance. R<sup>AB</sup> means the end-to-end resistance. R<sup>W</sup> is the wiper resistance which is typically 75 Ohm.

tunjid, 2015-02-20:
I'm aware of the ideal equation, what I'm saying is that for the calibration to be effective we need the real values of the digipot, not the ideal ones. 

We're working with the 1 K variant of the digipot and I've already read the factory calibrated values of the nominal values of the R_AB resistance of the EEPROM of the device.

R1 and R2 have different values for R_AB and R_W so we can't go off the equation on the data sheet. Each step can increase the digipot from anywhere between 14 and 18 ohms at a time, that's why we need a lookup table. I'm asking for the most accurate and efficient way to do that.

closed 2015-03-09 by tunjid

barrybelmont, 2015-03-09:
How did you solve this issue?