WuMRC/drive #24: Z_LoggerComplex readings are off

LKY21, 2017-03-27:
I am having issues measuring impedance using Z_LoggerComplex. My setup is pretty much the same as the one in the calibration [thread ](https://github.com/WuMRC/drive/issues/19)([fritzing](https://cloud.githubusercontent.com/assets/315476/5593475/1cb6b548-91c6-11e4-93cf-0756570272ef.png)) but with a 5934 and 16 Mhz coming from a funcgen. I've also commented out the temperature read portions. After attaching 98kohm resistors to Rfb and Rcal, I went through the calibration procedure (pressing p and enter twice). The serial monitor reported a gain factor of 17690708.00 and a phase shift of 2.00. Here's a quick compilation of results with a variety of resistors:

Ractual: Zmeasured
inf (no connection): 100000
940k: 18100
98k: 1844
21.3k:1246
1.028k: 1080
324 ohm: 1081
10 ohm: 1081

Looks like I'm not calibrating correctly. How should it be done? 
I've gone through Analog's CN-0217 document but nothing odd sticks out to me. Any other issues with my setup?

tunjid, 2017-03-28:
@LKY21 without using an analog front end for the AD5933, you're not going to be able to read values in the low hundred Ohm range. You should be able to read values in the 1K to 10k range though, but again that depends on your choice for Rfb. Refer to the the [Ad5993 Datasheet](http://www.analog.com/media/en/technical-documentation/data-sheets/AD5933.pdf) for some more info on the chips capabilities and Rfb selection criteria.

LKY21, 2017-03-28:
I understand that the range is 1k-10M so it makes sense that the readings bottom out at ~1080. I won't need to take measurements <1kohm for the final application, so can I do without the analog front end?

 It looks like the upper end isn't calibrated correctly either. Is the gain factor within reason?

tunjid, 2017-03-28:
It's a combination of a couple things: your choice of Rfb, the range (1 -4) you've set the AD5933 in, and the PGA gain you've chosen (x1 or x5). The library lets you chooses any of the 4 ranges and any of the 2 PGA gain values. 

Depending on the range of impedance you want to measure, you will need to pick the optimum combination of these 3 variables. This [application note](http://www.analog.com/media/en/technical-documentation/application-notes/AN-1252.pdf) should be exactly what you need for choosing all the values you need. It also mentions choosing the settling cycle for measuring, an option which is also provided by the library.

LKY21, 2017-03-29:
Great, thank you! I was looking for a document like this. Numbers come out properly now. 

I just came across the comment in the library about the gain factor being reciprocal to that in the datasheet. Makes sense now, thanks!